CAP1Sup/Intellistep #11: flashes, reads "Oled Init... OK:" and no operation

RocksaltX, 2021-04-08:
Hello, I have compiled this and loaded, but as title states. Did I miss something?

CAP1Sup, 2021-04-08:
> Hello, I have compiled this and loaded, but as title states. Did I miss something?

Nope, you're all good. This is a known issue. The encoder is unable to communicate, so it locks the entire system. Sorry about that, I'll hopefully be fixing the problem later today.

RocksaltX, 2021-04-08:
Another thing to note is that when received, the S42BV2 has all dips off and is closed loop. Once the firmware is changed the dips are backwards and closed loop does not work anymore. I did get a flash to work once, but on the printer the Y axis would move properly one way but not the other(shudder). Also, when I push the axis over say 200mm the motor would cut off.
Both files I saved from eeprom show same data, but each is different file size..
Is there a way to clone the eeprom as was from factory?

CAP1Sup, 2021-04-08:
@RocksaltX 
There is a copy of the production firmware in the precompiled folder. That will reset it back to factory firmware. Make sure to calibrate it afterward.
I would assume that the newer firmware that you're referring to is the BTT version? I have no control over that code. All that I know is that the code that I have written should work properly. All the issues you mentioned have been looked over and corrected in this rewrite. We're just stuck on the encoder. After that, we'll be able to check the stepper calculations and have the driver functioning. You've just caught us with our pants down (in development).

closed 2021-04-11 by CAP1Sup