CAP1Sup/Intellistep #16: Question on "Precompiled" files

jubjub64, 2021-05-04:
What is the difference between firmware-v2.0.bin and v2ProductionFirmware.bin found [here](https://github.com/CAP1Sup/Intellistep/tree/master/precompiled) ? v2ProductionFirmware.bin seems to be a much larger file so I wasn't sure which of these two was suitable for my S42B v2.0

CAP1Sup, 2021-05-04:
The `firmware-2.0.bin` is the one that BTT provided. The `v2ProductionFirmware.bin` was stripped from the flash memory of an actual production unit. This is essentially a factory reset. You want the production firmware. Make sure to recalibrate after flashing. I'll make a note later this week so that everyone understands what they mean.

jubjub64, 2021-05-04:
I see, thanks!

closed 2021-05-04 by jubjub64