CAP1Sup/Intellistep #22: S42B, S57B schematic

IhorNehrutsa, 2021-06-02:
Are there electrical schematic diagram of S42B, S57B?
Has anyone spent reverse engineering?

CAP1Sup, 2021-06-02:
Here is a rough schematic from @jlmxyz

I'll include this image and the schematic files in the repo soon. Could you please create a pull request with the schematics @jlmxyz ?

https://github.com/CAP1Sup/Intellistep/discussions/19#discussioncomment-770117

IhorNehrutsa, 2021-06-03:
Dear @CAP1Sup 
I tryed compile the betterReverse branch. I get firmware.bin size 57008 bytes, but it does'n works.
I am interested for G91, G90, G0, G1 with steps position absolute and incremental commands.
Also I am ready to testing on hardware.
Thanks for your time.

CAP1Sup, 2021-06-03:
Unfortunately, I haven't had any actual testing time recently, so most of the code written on the betterReverse branch is untested. To be completely honest, I would have been surprised if it actually worked the first time. I have something that I need to finish up for this weekend, but afterwards I will debug the issues and will get everything working, hopefully soon. 

As for the stepping commands, that's going to be more tricky. I don't want to play around with managing the timers and making the motor step right now, so that's either a feature that you'll have to wait a long time for or you could develop your own and pull request it. Unfortunately, the stepping isn't fully working (top priority, will be fixed soon), so it would just be theoretical, but having the groundwork to control speed and acceleration would be nice. The implementation in the parser is easy, it's the precise calling of the step routine that would be complex.

jlmxyz, 2021-06-05:
> Here is a rough schematic from @jlmxyz
> 
> [#19 (reply in thread)](https://github.com/CAP1Sup/Intellistep/discussions/19#discussioncomment-770117)
> 
> I'll include this image and the schematic files in the repo soon. Could you please create a pull request with the schematics @jlmxyz ?

I completed the schematic, there is still things to check (in particular, don´t know how CAN_EN signal works... and I still hadn't identified the 24v->5V (U8) chip)

still no values (I can't know capacities, neither power diodes or selfs)

the pull request draft : 
https://github.com/CAP1Sup/Intellistep/pull/24

CAP1Sup, 2021-06-05:
@jlmxyz I'll merge it now. Let me know if there's any changes, I'd like to keep it as up to date as possible.

CAP1Sup, 2021-06-09:
@jlmxyz I think that your diagram is incorrect. The stepping pin is PA-0, and the direction pin is PA_1. Please update it in your schematics and pull request the changes. Thanks!

IhorNehrutsa, 2021-06-16:
@CAP1Sup Please would you publish the latest(daily) current precompiled firmware. I want to compare it with my compiled bin file.

CAP1Sup, 2021-06-16:
> @CAP1Sup Please would you publish the latest(daily) current precompiled firmware. I want to compare it with my compiled bin file.

Why would you need it? Just curious. The latest development is in dev, the las test stable in master. It should all compile the same. What issues do you have so far?

IhorNehrutsa, 2021-07-02:
Replace PA0 and PA1 lines  
[image]

CAP1Sup, 2021-07-02:
Why? Aren't they wired correctly?

IhorNehrutsa, 2021-07-02:
https://github.com/CAP1Sup/Intellistep/blob/67add61c2d3b767a536940178ca9ce69527c692b/src/user/config.h#L248-L251
PA0 is the STEP_IN and must be connected to pin 6 of U3

CAP1Sup, 2021-07-02:
You're right! I can't believe I missed that. @jlmxyz could you fix the schematics?

CAP1Sup, 2021-07-13:
@IhorNehrutsa I forgot to say, but I fixed the schematics

jlmxyz, 2021-08-07:
sorry, lot of stuff to do here (work reconversion) I'll re-check on HW because it's strange... I trippled checked measurements... but I might have done a mistake....
the schematic will be updated, allmost everything is complete, appart capacitor and resistor values, and I started a pcb re-routing with components at right place (for upgraded version with enhanced functions)

didn't had time to test the firmware too... is there any prebuilt that i can put fast on the controler to perform tests?

jlmxyz, 2021-08-13:
I'm fixing the merge issues... and I noticed you moved the folder name from S57B to S42B.... the reverse Ing is done on S57B since it has CAN support that don't have the S42B.... so I think it's better to keep the schematic as S57B since it's closer to reality... a reverse should be done on S42B to see the differences with S57B, I expect only CAN is missing on S42B but I must check

CAP1Sup, 2021-08-14:
> I'm fixing the merge issues... and I noticed you moved the folder name from S57B to S42B.... the reverse Ing is done on S57B since it has CAN support that don't have the S42B.... so I think it's [...]

I assumed you were basing it off of an S42B V2. The V2 has CAN, just like the S57B

jlmxyz, 2021-08-14:
> I assumed you were basing it off of an S42B V2. The V2 has CAN, just like the S57B

hum sorry, I don´t have S42B v2 hw... seems that my S42B board is v1 since no CAN... so I can't tell the differences between both... I assume that both should be the same from software point of view, but since I can't double check the schematics against a S42Bv2 board....

CAP1Sup, 2021-08-14:
Fair enough. I think BTT literally just copied the S42B V2 designs for the S57B. The only difference seems to be a larger PCB, but the components haven't even been moved

IhorNehrutsa, 2021-08-14:
I can take pictures(photo) of the  S42A, S42B V2, S57B

jlmxyz, 2021-08-28:
Ok I got a S42Bv2, sadely the schematics will have to be updated for this board... first fast look show at least component name change, component size and value change (on power resistor at least), and some components are not present (crystal....)

by the way, the firmware will have to be updated for s57bv2 since there is an external cristal....

StefanBruens, 2022-05-10:
The DC converter is an A&O AOZ1282CI, see last page for package marking (BN <OPTION> <WEEK>).
http://www.aosmd.com/res/data_sheets/AOZ1282CI.pdf

CAP1Sup, 2022-05-10:
Do you mind editing the PCB files to match? I will be happy to merge a PR