CAP1Sup/Intellistep #25: S42B doesn't reverse at endstop switch

jubjub64, 2021-06-16:
I couldn't set the stepper direction through the lcd interface so I just reversed my steps in marlin using the M92 command. I read that the S42B was set to 1/16 so I set it to -80 for x and y (which is where my S42Bs are located). They were quiet until I actually moved them, then they make a continuous high pitched noise, but they still moved fine. It was when I homed my printer that the real problem occurred, it wouldn't detect my endstop switch and just started to grind into it with an aggresive vibration.

CAP1Sup, 2021-06-16:
Hmm... I'll have to look into setting the inversion on the OLED. That should have worked fine. I'll add this issue to the TODO list.

The behavior from quiet to loud is normal. When the motor is disabled, the coils are set to the idle mode to reduce noise. I am currently focusing on other features right now, but making the stepper quieter will be a priority soon.

As for the grinding, I believe that the issue is that the mainboard stopped commanding the stepper, yet the board was set to go farther. I'm going to push a minor update to the main branch to fix an issue that I saw in the stepping logic. Would you be willing to test the motion without the step correction enabled? This can be done by turning your third dip switch from the left off (as long as your board isn't inverted). That should solve the aggressively clicking issue, but the motor will just act like any old "dumb" motor. It would be good to test it just so I can get an idea of if it needs fixed.

jubjub64, 2021-06-17:
Ok, here are some videos of performance with closed-loop disabled.  

Here is me moving back and forth on the X-axis. It seems to keep going in one direction until I give it a new command even though it was only supposed to go 10mm.
https://youtu.be/ruNul-CXP7s

Here is the autohome function video. There is this high pitched noise like the sound effect of a bomb dropping in an 8-bit game. Kinda scraed me!
https://youtu.be/Ka_fgUyH6jY

CAP1Sup, 2021-06-17:
Wow, what excellent performance.... Sorry that you ran into those issues. 
Before we get down to checking what the final issue is, could you check that you have the following in your Marlin Configuration_adv.h?
`#define MINIMUM_STEPPER_PULSE 3
#define MAXIMUM_STEPPER_RATE 150000`

jubjub64, 2021-06-17:
Well, I understand that I'm helping test so I'm not dissapointed. I do have #define MINIMUM_STEPPER_PULSE 3
#define MAXIMUM_STEPPER_RATE 150000 set already.

CAP1Sup, 2021-06-17:
@jubjub64 I think I know why it's doing that. Try turning both of your left side dips on. I believe that the motor is in full step mode instead of 1/32 (should be that because I set a microstructure multiplier of 2, making the overall fraction 2/32, or 1/16). Make sure your dips are the right way around as well.

CAP1Sup, 2021-06-17:
@jubjub64 commit 6cd614600365f91cf7a0dccf65392e630a3fe15c fixes your issue with reversing the motor. I want to add parameter settings to the main menu, then I will join the dev branch with the master for everyone to see the code

closed 2021-06-17 by CAP1Sup