CAP1Sup/Intellistep #30: Calibration required to start stepper but can't calibrate

jubjub64, 2021-06-19:
Displays says to calibrate but when I try to calibrate it says the feature is not implemented yet. I don't seem to be able to use the motor until it is calibrated. Even when closed loop is disabled it insists on calibrating first and won't let the motor move.

jubjub64, 2021-06-19:
This is with both the master and dev branches currently.

CAP1Sup, 2021-06-19:
> This is with both the master and dev branches currently.

Yup, just click the select button (the top) and it'll go through. I just left it in as a reminder that I need to add it eventually. As of right now, it's not needed, but later I want to implement a correction system that takes into account if the motor is actually moving

jubjub64, 2021-06-20:
So if I press the select button (top one) after it boots up it says that calibration isn't implemented yet and then reboots the s42b. If I select the middle button I can get the page that has the angle and temp readings to show but I can't get to the settings menu no matter what buttons I push. Either way, if I bypass the calibration screen to show the status page the printer will not move on the x or y axis where i have the S42B. I made a crummy video to show what I did to try and make it work. Sorry, filming this while my camera was facing upwards and still working the S42B buttons was tricky so the video is bad.
https://youtu.be/EtZHqN8TuYU

CAP1Sup, 2021-06-20:
No problem, the video was really helpful. It appears to be an issue with the writeFlash function not working properly. Normally, the calibration procedure (yet to be written) would run, then the calibration flash would be written. The fact that you can use the back button is actually a programming oversight, it was never intended to work. I'll update the code, just give me a minute.

CAP1Sup, 2021-06-20:
78c503d5314320d783dc53692051b1177ee4f222 should fix the issue with the motor not "calibrating". I think that it's an issue with the page not being able to be written to, so I just set it so that the flash is erased before the calibration is written. This is nothing but a temporary fix though, as it will lead to all of the parameters being wiped on every calibration. As of right now, the calibration doesn't actually mean anything, so I'm going to work on it. However, if you are still having issues, try disabling the check for the flash address in isCalibrated() (in flash.cpp) with just a true statement. That way, you'll be able to run your motors. If you do just change it, I'll take a look at this issue further as it is important to be able to save the parameters correctly.

closed 2021-07-07 by CAP1Sup