CAP1Sup/Intellistep #31: Driver speed

IhorNehrutsa, 2021-06-22:
BIGTREETECH declared the Product Parameters as:
```
Maximum speed: 1000RPM (revolutions per minute) 
Subdivision: 2 4 8 16 32
```
Stepper motor specifications is:
```
200 full steps per revolution (1.8° per step)
```
1000RPM is 1000/60 = 16.6666666667RPS (revolutions per second) 
When 32 Subdivision is used than MAX frequency of input STEP pulses must be:
16.7 * 200 * 32 = 106880 Hz

I has tested frequency of input STEP pulses with original BIGTREETECH firmware. 
It is near 50kHz.

I try to  check MAX frequency of input STEP pulses in dev branch.
I use red LED to see duty cycle of motor step
Intellistep\src\hardware\timers.cpp: 
```
// Just a simple stepping function. Interrupt functions can't be instance methods
void stepMotor() {
    digitalWriteFast(LED_PIN, HIGH);
    motor.step();
    digitalWriteFast(LED_PIN, LOW);
}
```
4kHz input STEP pulses:  
[image]  

red LED  
[image]  

5kHz input STEP pulses:  
[image]  

red LED  
[image]  

There is no more time to process input STEP pulses.
We need faster algorithms.

CAP1Sup, 2021-06-22:
@IhorNehrutsa thanks for doing the testing! Unfortunately, I don't have the equipment to do that kind of testing, so it's great to see how everything is working. Do you have any ideas on how we can speed up the stepping routine?

CAP1Sup, 2021-06-22:
Did you disable all of the closed loop interrupts and everything? That could be really hurting the performance

CAP1Sup, 2021-06-23:
@IhorNehrutsa since you have an oscilloscope, would it be possible for you to test the frequency of the oscillator on the board? I was unable to find a solid value that was actually accurate for the external (HSE), so I was forced to use the internal oscillator ￼(HSI). The thing is, when you use the internal, it forces you to cut the frequency down by half. This leads to longer compute times and slower overall performance. I think that we could get around 2x more performance by just figuring out that value

IhorNehrutsa, 2021-06-23:
HSI and PLL allow to set 64MHz SYSCLK. See STM32Cube
[image]

I measured HSE oscillator. It is 8MHz.

IhorNehrutsa, 2021-06-24:
> Did you disable all of the closed loop interrupts and everything? That could be really hurting the performance
Yes, I switch off correctionTimer in timers.cpp

CAP1Sup, 2021-06-29:
@IhorNehrutsa how are we doing on max update rate now? I'm thinking that it should be higher

IhorNehrutsa, 2021-06-29:
I will mesute it tomorrow in office.
My time zone is UTC+3.

IhorNehrutsa, 2021-06-30:
Not as I expected.
It's still 30kHz.
WIP

IhorNehrutsa, 2021-06-30:
with merged 
https://github.com/CAP1Sup/Intellistep/tree/Stepping-Speed-Improvements
47kHz
WIP

IhorNehrutsa, 2021-07-02:
Measured on master branch
https://github.com/CAP1Sup/Intellistep/commit/c014991977ff657ef11a74511b7668306258c99c
48.3kHz

https://github.com/CAP1Sup/Intellistep/pull/50
SINE_MAX a power of 2 to fast division to SINE_MAX
49.3kHz

CAP1Sup, 2021-07-02:
Truly a massive performance gain. Thanks so much Ihor. It was a great help! Can I close this now? I think we meet BTT's performance at this point

IhorNehrutsa, 2021-07-02:
1000RPM is 1000/60 = 16.6666666667RPS (revolutions per second)
When 32 Subdivision is used than MAX frequency of input STEP pulses must be:
16.7 * 200 * 32 = 106880 Hz
16.7 * 200 * 16 = 53440 Hz
16.7 * 200 * 8 = 26720 Hz

It is too early, 16 and 32 step subdivisions are still unachievable. :(

CAP1Sup, 2021-07-02:
If I do the calculations the other way, the maximum rate is 7.7 rps at 1/32 microstepping. Now, if you consider that a typical printer pulley is about 10 mm to 15mm in diameter, that would lead to a maximum linear speed of 242 mm/s and 363 mm/s respectively. I really don't think that people are going to be exceeding that. Also, I don't think that there's much more performance to be had unfortunately. Maybe some minor gains, but I don't think that we'll double our current rate

IhorNehrutsa, 2021-07-02:
My target project is not a 3D printer. It is a high-speed tracking/positioning system. 
1/8 step is too loud and twitchy.

IhorNehrutsa, 2021-07-02:
How to force the compiler to save preprocessed files?
I can't find it.
Thanks.

CAP1Sup, 2021-07-02:
> My target project is not a 3D printer. It is a high-speed tracking/positioning system.
> 1/8 step is too loud and twitchy.

Sorry for assuming that you were 3D printing, I just expected that most people would be using this for a printer. It looks like you were able to find out how to save preprocessed files, which is good because I didn't. We'll have to keep on working on the stepping routine to continue to increase the maximum speed.

xerootg, 2021-07-02:
To add to this, I'm running 2mm pitch leadscrews. I need fast, much faster
than the stock firmware which runs out of steam at 660rpm.

CAP1Sup, 2021-07-05:
@IhorNehrutsa now I know that STM32CUBEMX says that 72Mhz is the max for the SYSCLK..... but considering that we need more performance and programming optimizations are getting harder.... I did some testing and was able to increase the SYSCLK to 128MHz while keeping the system stable for functions such as Serial and stepping. Could you give the dev branch a quick test? I'm thinking that the stepping rate should increase to around 75-100kHz.

IhorNehrutsa, 2021-07-06:
53.5kHz

CAP1Sup, 2021-07-06:
Darn! Do you have any idea why the speed really didn't increase? Maybe it's the register setting for setting pins? Also, I got PID somewhat working. I really don't like the library and I think that I'm just going to write the math into the PID class itself. That way, we can use floats to save time and will have much more control over the process

IhorNehrutsa, 2021-07-06:
I measure PLLCLK on MCO
https://github.com/CAP1Sup/Intellistep/pull/54

When SystemClock_Config_HSE_8M_SYSCLK_128M() is selected
PLLCLK is 128kHz

Stepping rate is 93kHz

Maybe I was inaccurate in the measurements in first time.
Maybe MCU started in other CLK way in first time.

CAP1Sup, 2021-07-06:
Getting really close to hitting the desired 100 kHz that you wanted. You can definitely run at 1/16th stepping. I'll look at increasing the GPIO clocks to see if we can squeeze more performance out of the processor

IhorNehrutsa, 2021-07-06:
My bad.
53.5kHz 
It was at dev branch with the SystemClock_Config_HSE_8M_SYSCLK_72M()

IhorNehrutsa, 2021-07-06:
55-56kHz
Speedup to 55kHz  with the SystemClock_Config_HSE_8M_SYSCLK_72M()  #55

Now I think we need to focus on the core functionality of Intellistep (PID + SimpleFOC SimpleFOCStudio mode), but keep in mind the speed.

IhorNehrutsa, 2021-07-06:
96kHz
Merge pull request #55 
https://github.com/CAP1Sup/Intellistep/commit/e5d39079f335ffcbe2f240c2e3fa608f9cc17ee5

CAP1Sup, 2021-07-06:
Wow, getting really close to the 101 kHz. Close enough for now. I'll work on PID today, hopefully get it working by tonight

IhorNehrutsa, 2021-07-13:
Revert up to 55kHz #73
with the SystemClock_Config_HSE_8M_SYSCLK_72M()

CAP1Sup, 2021-07-13:
Is it slower with the pow() function?

IhorNehrutsa, 2021-07-13:
Yes, I took 2 days to find it.

CAP1Sup, 2021-07-13:
Sorry about that. I thought that the compiler would optimize it. We'll have to be more careful in the future

IhorNehrutsa, 2021-07-15:
Encoder improvements and Fix microstepMultiplier #78
59kHz  SystemClock_Config_HSE_8M_SYSCLK_72M()
102kHz SystemClock_Config_HSE_8M_SYSCLK_128M()

CAP1Sup, 2021-07-15:
We finally broke 100kHz!

IhorNehrutsa, 2021-07-16:
Encoder improvements and Fix microstepMultiplier #78
102kHz   SystemClock_Config_HSE_8M_SYSCLK_128M()

IhorNehrutsa, 2021-08-04:
Step-dir under ideal conditions #83
87.6kHz  SystemClock_Config_HSE_8M_SYSCLK_72M()
150kHz   SystemClock_Config_HSE_8M_SYSCLK_128M()

CAP1Sup, 2021-08-04:
No way! 150kHz is insane

IhorNehrutsa, 2021-08-04:
Remove CLOCKWISE and COUNTER_CLOCKWISE
91.2kHz SystemClock_Config_HSE_8M_SYSCLK_72M()
150.3kHz SystemClock_Config_HSE_8M_SYSCLK_128M()

CAP1Sup, 2021-08-04:
Don't we need CW/CCW to specify rotation direction?

IhorNehrutsa, 2021-08-10:
Free motor shaft, smooth acceleration, ERR grows smoothly. 
[image]

CAP1Sup, 2021-08-10:
What exactly are you seeing?

IhorNehrutsa, 2021-08-11:
[video](https://drive.google.com/file/d/1k-W4RpAqn1isayHsZECJWy0fMSj7Up2_/view?usp=sharing)
I expected to see something near the black graph(my Vmotor=11V and Imotor=500mA) from 
[NEMA 17 Stepper Motor](https://github.com/CAP1Sup/Intellistep/blob/master/datasheets/nema17.pdf)
[image]
The experiment seems to me satisfactory.