CAP1Sup/Intellistep #35: PID

IhorNehrutsa, 2021-06-24:
I propose to move out the PID controller from the StepperMotor class to separate class and files.
The PID controller is not a property/characteristic/attribute of a motor.
This simplifies the StepperMotor class and allows to use of a third party PID library.

It may be 
https://github.com/br3ttb/Arduino-PID-Library
with detailed explanation here
http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-introduction/

There is also a PID auto-tuning library
https://github.com/br3ttb/Arduino-PID-AutoTune-Library

CAP1Sup, 2021-06-24:
I started working on this, but I wasn't able to finish it today. I'll publish the commit within the next couple of days.

CAP1Sup, 2021-06-24:
52c3023e47f1f1cc47eab8ed8c8430d066abb496 makes PID it's own class. I haven't implemented it yet, but that will be soon. My plan is to use the PID loop to control the update time for an interrupt timer. Thoughts on that or any of the commits so far?

CAP1Sup, 2021-06-29:
I'm getting close to getting PID fully working, I should finish it tomorrow or Wednesday if everything goes smoothly. Autotuning will come later, but at least you'll be able to use the basic manual tuning.

CAP1Sup, 2021-06-30:
I'm just going to move everything to the SimpleFOC library. I'm going to create a separate branch for it, so make sure that you work on that branch if you wanted to make any changes (it turns into a nightmare) and PR the SimpleFOC branch

xerootg, 2021-06-30:
I have a basic simplefoc implementation on my GitHub. It's using the wrong
pwm implementation, but if you are interested, I can push my local branch.

CAP1Sup, 2021-06-30:
@xerootg that would be terrific! Could you pull the latest commits on the dev branch, then create a pull request on it? That would allow me to work with you to bring the functionality to the Intellistep. It would certainly be interesting to have it.

@IhorNehrutsa let me know, I'll give you perms to review PRs, that way we can all work on it

CAP1Sup, 2021-06-30:
@xerootg is the motor quieter with SimpleFOC? I'm curious as to if it's just better to look at the SimpleFOC code and ~~steal~~ reference their code. I think that my PID implementation is on the horizon of working, so it might actually just be better to stick with what we have already and adapt some of the code

xerootg, 2021-07-01:
I wouldn't say that I have tuned the driver very well, as I wrote just enough code to make it spin. It needs some tuning in regards to base frequency and I haven't quite wrapped my head around their model for duty cycle in relation to torque.

CAP1Sup, 2021-07-06:
@IhorNehrutsa commit 8396fe7c48c6f8134665810efd8732a380d94225 implements custom PID. It's really fast and I had a good amount of success with it. Those values should get you started, although I won't pretend that I know what I'm doing. I also added the start of the Python tuning utility.