CAP1Sup/Intellistep #41: It was uint32_t before

IhorNehrutsa, 2021-06-30:
https://github.com/CAP1Sup/Intellistep/blob/adb9a8e7c835396826516a921b5bfd0fd209e1cc/src/hardware/flash.cpp#L269
https://github.com/CAP1Sup/Intellistep/blob/adb9a8e7c835396826516a921b5bfd0fd209e1cc/src/hardware/flash.cpp#L271
https://github.com/CAP1Sup/Intellistep/blob/adb9a8e7c835396826516a921b5bfd0fd209e1cc/src/hardware/flash.cpp#L333

CAP1Sup, 2021-06-30:
I wanted the ability to set a negative number to show that the index was invalid. Float is the only negative data type that is supported by the functions and I didn't think it was necessary to create a int32_t function.

IhorNehrutsa, 2021-06-30:
I am not sure about  the type cast  (AXIS_CAN_ID)(-1.0) here
```
setCANID((AXIS_CAN_ID)readFlashFloat(CAN_ID_INDEX)); 
```

CAP1Sup, 2021-06-30:
I guess we could filter the -1 then

CAP1Sup, 2021-06-30:
Fixed in commit https://github.com/CAP1Sup/Intellistep/commit/8f6ffb952e84fc41830f03430d908f1fa2a6ac7d

commit a506b0e6 (2021-06-30): Update flash.cpp

Fix #41
closed 2021-06-30 by CAP1Sup