CAP1Sup/Intellistep #46: Microstep multiplier

IhorNehrutsa, 2021-07-01:
There is the microstepDivisor that sets the correspondence between fullStepAngle and microstepAngle
https://github.com/CAP1Sup/Intellistep/blob/c014991977ff657ef11a74511b7668306258c99c/src/hardware/motor.h#L200-L207

There is the microstepMultiplier, but it does not related to the microstepDivisor. What is its physical meaning?
https://github.com/CAP1Sup/Intellistep/blob/c014991977ff657ef11a74511b7668306258c99c/src/hardware/motor.h#L220-L221
MICROSTEP_MULTIPLIER defined as 2.

CAP1Sup, 2021-07-01:
So the microstep divisor is the number on the bottom of the fraction of the microstepping. For example, it would be 16 if we were using 1/16 microstepping. The microstep multiplier is the number of microsteps to move per step. This allows the motor to pretend to run in (in this case) 1/16, while really correcting in 1/32. Think of correction steps like:
1/microstepDivisor

and steps per the step pin as:
microstepMultiplier / microstepDivisor

CAP1Sup, 2021-07-02:
I'll add documentation to the dev branch, it'll push into the master soon

closed 2021-07-02 by CAP1Sup