CAP1Sup/Intellistep #56: Possible Missed Steps

CAP1Sup, 2021-07-07:
I was thinking about the interrupts on the stepper board and realized that there could potentially be a problem. Currently, the encoder and some other time sensitive functions disable interrupts while running. This wouldn't be a problem, except for the fact that there cannot be a single step skip. What happens if there is a step pulse during a function that disables interrupts? If a step pulse comes in, will the interrupt process once the interrupts are re-enabled? We need to ensure that no steps are lost, because otherwise, the board is useless.
@IhorNehrutsa could you take a look at this?

IhorNehrutsa, 2021-07-07:
I don’t know yet how to reliably check this.

In any case, interrupt routines must be kept as short as possible.

CAP1Sup, 2021-07-07:
Hmm... the way that the board usually works is that the step pin sets the pending bit for the interrupt. Upon seeing that, the CPU stops what it's doing and executes the code. Therefore, I think that if we made a function that couldn't be interrupted, then we could add a delay, set the pending bit, then delay a little more, and finally re-enable interrupts. That would give us an idea of how the interrupt would be processed. Now I don't know if the disabling of interrupts prevents the pin from setting the pending bit or the pending bit is just ignored if the interrupt isn't enabled. I'll look into it later today

CAP1Sup, 2021-07-08:
@TazerReloaded I know it's been a while, but you're the best with the encoder. Is there any way that we can turn the SPI transfers to be uninterruptable without disabling interrupts? I tried HAL_SPI_TransferReceive_IT, but I wasn't able to get it to work right. Could you look into this?

CAP1Sup, 2021-07-09:
I was right. It causes minor layer shifts over the part. Darn

CAP1Sup, 2021-07-09:
[image]

CAP1Sup, 2021-07-09:
The thing is that it's not an issue with the PID or something like that, it's an encoder issue (I think). The flash is the only other file that calls the disableInterrupts, but that only runs while modifying the stored parameters. Encoder is what is causing the layer shift. I think that the best way to deal with it would be to make the encoder into an interrupt with an extremely high priority, then have a function that sets the interrupt to run.

xerootg, 2021-07-12:
Totally naïve question here; is the CNT attribute of a timer also dead when interrupts are disabled? I remember working on a project where I moved my pulse counter to one of the hardware peripherals to avoid lost pulses.

IhorNehrutsa, 2021-07-12:
@xerootg I have looked for a hardware peripheral counter for Stepping. 
It seems not possible in this case.

IhorNehrutsa, 2021-07-12:
I'm pretty sure there is no need to disable interrupts when reading encoder registers. I will test this tomorrow.

xerootg, 2021-07-12:
> @xerootg I have looked for a hardware peripheral counter for Stepping.
> It seems not possible in this case.

Would you happen to remember why? The datasheet says that tim1 and tim8 are like the perfect timers for doing counting across multiple inputs, but I know the datasheet lives in an idealistic world even before accounting for ST's eratta sheet and I personally have never quite figured out how to use them 🤣

IhorNehrutsa, 2021-07-12:
In our case, we can use only the TIM2 and the PA0 pin.
 And I don't find a use case for a hardware counter.

CAP1Sup, 2021-07-12:
I think that BTT might have used a hardware counter. The issue is that the steps are not executed when desired, making accelerations inconsistent. Also, there is a need to disable interrupts while reading encoder registers. I think from testing I found out that the values are unstable due to occasional misreads. This would obviously be an issue with PID.

xerootg, 2021-07-13:
BTT did indeed use tim2 - [here](https://github.com/bigtreetech/BIGTREETECH-Stepper-Motor-Driver/blob/04f9fb9a646be17448d51e7006858570ae58e68d/S42B/firmware/S42BV2.0/Close_loop/src/hardware/time.c). tim2 is particularly interesting, it has step (tim2_ch1/tim2_ch1_etr), dir (tim2_ch2) and en (tim2_ch3) all three as separate channels, and since tim2 is a general purpose timer, it also supports what ST calls 'encoder mode' supports hardware influenced direction for the counter in addition to overflow and underflow detection; I'd hook the en signal up to the enable flag for the timer to keep the timer from counting, use ch2 as the direction. Encoders do this with literal time, I can't imagine that a state at the point in which the input is called would be even as complex as what the encoder mode is often used for. Its also a 32 bit counter (which can be DMA backed!) as well as providing an interrupt and a change bit that can be consumed in another timer, such as the PID loop.  dir and en are also backed by tim5, another general purpose timer, which could enable a bit more flexibility since timers can be chained.

In addition to all of that, there's a filter for signal transitions in front of the timer that allows for filtering noisy transitions, a.k.a. misreads. ST planned for this to be used for things like positional command in real-time with minimal impact on the main loop.

If I was doing more than armchair development here, I'd be implementing encoder mode on tim2 to ensure that 1) steps aren't lost and 2) that it happens in a timely manner that does not starve the loop when no change in commanded position has been requested. I keep saying, 'oh yeah, I can take a look at that arbitrary problem' and then getting randomized by life, so I'm not gonna volunteer to do anything else but I'd love to review anything involving the timer, if for nothing else, so that I can contribute something meaningful and learn something.

CAP1Sup, 2021-07-13:
@xerootg is it possible to attach the timer and an interrupt function at the same time? Then we could use a step interrupt to move the motor, with the PID loop referencing the master TIM2

CAP1Sup, 2021-07-14:
@xerootg I managed to get the hardware counter working. The only thing is that the timer interrupts are not set up for overflows.

xerootg, 2021-07-14:
Oh, that's wonderful! I was gonna spend some time in the manual tonight to answer your prior question, sounds like you have a path forward? The best documentation I've found so far has been literally school assignments from random EE classes. The notes in some of those lectures are amazing. It seems like 'tim2' is consistent in features across the current generation of STM32's, and the 407 seems to be extremely popular in some of the lecture notes I found when I was trying to figure out how encoder mode worked.

CAP1Sup, 2021-07-14:
Yup, I was able to step the motor and count it at the same time

CAP1Sup, 2021-07-15:
@xerootg 8abd653 and its parent, 75b10a3, finish the implementation for the hardware step counter. Thanks for the idea. This is an excellent idea to help to prevent missed steps. Please let me know if you have any other ideas on how to improve the performance of the boards.

closed 2021-07-15 by CAP1Sup