CAP1Sup/Intellistep #77: If disable CAN then encoder is blocked

IhorNehrutsa, 2021-07-15:
To reproduce the error, comment out the line 
https://github.com/CAP1Sup/Intellistep/blob/a8546f44709449c86ac5067fbde60f9c229bf54a/src/user/config.h#L62

It seems is bug here 
https://github.com/CAP1Sup/Intellistep/blob/a8546f44709449c86ac5067fbde60f9c229bf54a/src/lib/eXoCAN/eXoCAN.cpp#L95-L105

CAP1Sup, 2021-08-03:
Please confirm that this is fixed, then I will close the issue

CAP1Sup, 2021-08-15:
@IhorNehrutsa is this fixed? I haven't had a chance to test it myself.

IhorNehrutsa, 2021-08-16:
Yes, it works

CAP1Sup, 2021-11-28:
I was doing some testing and disabled the CAN functionality. I was unable to get the encoder to send values. I re-enabled CAN and everything was good. No idea what changed, but this needs to be looked into again.

closed 2022-01-29 by CAP1Sup