CAP1Sup/Intellistep #79: Revert variable name 'softStepCNT' to 'desiredStep'

IhorNehrutsa, 2021-07-15:
https://github.com/CAP1Sup/Intellistep/blob/f292d5d595b82d5efa1879fca8f6c93d19482ed4/src/hardware/motor.h#L221-L222
The 'softStepCNT' was the 'desiredStep'

'desiredStep' expresses the purpose of a variable (essence, entity, nature).  
'softStepCNT' expresses how a variable is handled (function, method). 

The method may change.
At first, the method was stepping interrupt, now the method is TIM2(CNT), DMA (TIM2 (CNT)) is possible in the future.
But the 'entity' stays the same - the 'desiredStep'.

Renaming is not a difficult procedure in VSCode, but it may inconvenient, if you make big changes in potent branches. 
If you do not object to renaming, I may do it.

CAP1Sup, 2021-07-15:
It's not really the desired step count, it's more of a software counter. That's why I renamed it. The desired step is derived using the specified method, not just the software version.

CAP1Sup, 2021-07-15:
I don't think that it's even used anymore, now that the hardware counter is in place. Maybe we could get some minor performance benefits by removing it? It doesn't really make sense reflecting on it

xerootg, 2021-07-15:
I haven't had time to look at the tim2 implementation, but if you are
indeed referencing tim2->CNT then it sure seems like counting in software
is just lost cycles.

CAP1Sup, 2021-07-15:
Would you mind taking a look at it? You seem to know what you're doing. Myself? Not so much

IhorNehrutsa, 2021-07-15:
Are you sure that int32_t is enough?
https://github.com/IhorNehrutsa/Intellistep/blob/ad46de343e5d191641bd608baa850723b871c984/src/user/config.h#L17-L21
Maybe int64_t is required in the future.

Look here http://leadshine.com/series.aspx?type=products&category=stepper-products&producttype=stepper-drives&series=EM-S
The lowest model EM415S has resolution 200, 400, 800, 1600, 3200, 6400, **12800, 25600** steps
The highest model EM882S has incredible **40000** steps.

xerootg, 2021-07-15:
It's on my 'todo' list 🤣 I promise I will look.

CAP1Sup, 2021-07-15:
@IhorNehrutsa I really don't think that we need a 64 bit integer. Most users will never exceed over a hundred, let alone a thousand or more revolutions.

IhorNehrutsa, 2021-07-15:
If we stay on the in32_t size of the step counter, then softStepCNT is not needed.

CAP1Sup, 2021-07-15:
Sounds good to me

IhorNehrutsa, 2021-07-15:
Oh no, there are other interfaces to set 'desiredStep' in PID mode. It may be SET_POSITION command through the UART or CAN.

CAP1Sup, 2021-07-15:
The hardware counter can be set, I made a function for it

IhorNehrutsa, 2021-07-15:
Maybe _/0_/

IhorNehrutsa, 2021-07-15:
335544 rev's / 1000RPM ~ 336 min == 5.6 hour in one direction for counter overflow.
I vote for int32_t.

IhorNehrutsa, 2021-07-15:
I am confused by 'increments_t'. The first association is increment/decrement, not a 'step'.

CAP1Sup, 2021-07-17:
Sorry about that, I misunderstood. I adjusted the comments on your PR to make it more clear. I think that I can close this now, your solution was excellent

closed 2021-07-17 by CAP1Sup