CAP1Sup/Intellistep #91: Microsteps which settled through the menu is forgotten after the reset.

IhorNehrutsa, 2021-08-05:
After the reset, the DIP microsteps are setted again.

IhorNehrutsa, 2021-08-05:
The solution is to save the DIP state in flash and set new microsteps if DIP state is changed (even if DIP changed when driver was powered off).
Save microsteps in flash after menu or after M350 command or after DIP state is changed.
Read microsteps from flash after a reset.

CAP1Sup, 2021-08-05:
That's essentially what it does now, it's just that the settings need to be saved in order to persist. The flash values should override any DIP values. I think it's best to leave the saving to the user as it will prevent the accidental saving of unwanted settings

IhorNehrutsa, 2021-08-05:
>>  The flash values should override any DIP values. 

Disagree. The last changed value is the right value. If the user changes the DIP( even if DIP changed when the driver was powered off or after the menu or after M350), he wants that and expects that.

CAP1Sup, 2021-08-05:
> >>  The flash values should override any DIP values. 
> 
> 
> 
> Disagree. The last changed value is the right value. If the user changes the DIP( even if DIP changed when the driver was powered off or after the menu or after M350), he wants that and expects that.
> 
> The latest value is right.

Ok. Open a PR, we can add the checks so that the DIPs will be handled properly

CAP1Sup, 2021-09-13:
I think this is done, so I'm going to close it for now.

closed 2021-09-13 by CAP1Sup