CAP1Sup/Intellistep #95: Measure frequency(period) of input step pulses

IhorNehrutsa, 2021-08-06:
Is it possible to measure frequency(period) of input step pulses on fly by hardware counters?
This allow to continue correction timer with proper(last input step) frequency(period).

CAP1Sup, 2021-08-06:
I don't think so. I did a large amount of research into the timers and I don't remember ever coming across a period counter. We could do something with software though, it's not too hard, especially considering that everything is perfectly correct with the counter

xerootg, 2021-08-07:
Yeah, you should be able to measure durations in a timer. It's gonna be
dependent on what pin you want to count on though... Tim2 can definitely do
it, but that's currently counting pulses. It's not a huge jump to say that
frequency is pulses over time, so maybe that's a starting point? Do you
have a copy of the full manual for the stm32f1xx?