CAP1Sup/Intellistep #104: compilation error in main.cpp

jlmxyz, 2021-08-16:
Hi,
I tried to build intellistep, opened the project with vscode + platformio plugin,
I get the following error : 
```
Processing BTT_S42B_V2 (platform: ststm32; framework: arduino; board: genericSTM32F103CB)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103CB.html
PLATFORM: ST STM32 (11.0.0) > STM32F103CB (20k RAM. 128k Flash)
HARDWARE: STM32F103CBT6 72MHz, 20KB RAM, 128KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pioenvs/BTT_S42B_V2/src/user/main.cpp.o
src/user/main.cpp: In function 'void setup()':
src/user/main.cpp:21:5: error: 'overclock' was not declared in this scope
   21 |     overclock(RCC_CFGR_PLLMULL16);
      |     ^~~~~~~~~
src/user/main.cpp:28:5: error: 'initEncoder' was not declared in this scope; did you mean 'Encoder'?
   28 |     initEncoder();
      |     ^~~~~~~~~~~
      |     Encoder
src/user/main.cpp:31:22: error: no matching function for call to 'StepperMotor::enable(bool)'
   31 |     motor.enable(true);
      |                      ^
In file included from src/main/main.h:7,
                 from src/user/main.cpp:2:
src/hardware/motor.h:247:14: note: candidate: 'void StepperMotor::enable()'
  247 |         void enable();
      |              ^~~~~~
src/hardware/motor.h:247:14: note:   candidate expects 0 arguments, 1 provided
*** [.pioenvs/BTT_S42B_V2/src/user/main.cpp.o] Error 1
==================================================================================== [FAILED] Took 5.08 seconds ====================================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```

CAP1Sup, 2021-08-16:
Is this from the master, dev, or another branch?

jlmxyz, 2021-08-16:
dev, I miss a few commits, but diffs aren't related to the lines that cause
compilation errors

could be great to have build instructions.

CAP1Sup, 2021-08-16:
Let me see if I can build it for you, then I'll post the binary

CAP1Sup, 2021-08-16:
Darn, my work computer doesn't allow me to build the binary. @IhorNehrutsa can you help with this?

IhorNehrutsa, 2021-08-17:
to @jlmxyz 
It looks like not updated code

0) Go to your fork of Intellistep  https://github.com/jlmxyz/Intellistep
then Fetch and merge code from upstream
[image]

jlmxyz, 2021-08-21:
I'm not on master but on dev based branch : 
[image]

IhorNehrutsa, 2021-08-22:
The last picture looks ok.
Is Visual Code(PlatformIO) TERMINAL output stays the same as in first(top) message?

jlmxyz, 2021-08-23:
yes, I even rebased on latest, removed all .* that are created, tried
rebuild, same error....

closed 2021-09-03 by CAP1Sup