CAP1Sup/Intellistep #107: Incorrect saving of the configuration to memory

svsergo, 2021-09-02:
The writeFlash function is described under the same name several times.  After saving the configuration, it is read incorrectly.

CAP1Sup, 2021-09-02:
Could you explain what you mean further? The writeFlash function is overloaded, allowing it to execute different code based on the data type that is being stored. The readFlash functions could not be overloaded because C++ doesn't support overloading based on return types, leading to the renaming of the readFlash functions so that they have the data type at the end (ex readFlashU32).

svsergo, 2021-09-02:
This is an assumption that the error may be in this. After saving the configuration in flash, the information is no longer displayed on the screen. It is displayed again only after the Wipe operation.  
[image]

CAP1Sup, 2021-09-02:
Hmm... I'll take a look into it.

CAP1Sup, 2021-09-02:
@svsergo are you on the master or dev branch? (Default is master)

svsergo, 2021-09-02:
The firmware is loaded from the branch dev.

svsergo, 2021-09-02:
I checked the firmware from the master branch. The error is repeated.

CAP1Sup, 2021-09-02:
Excellent, that gives me a good idea of where I need to look. I haven't been able to test the recent updates as I've been busy and need the motors working at 100%, so I figured something like this would eventually happen. I'll get on it and hopefully have it solved by next week

CAP1Sup, 2021-09-02:
Alright, it was a quick fix! I somehow forgot to write the encoder offset when saving the parameters. The board went to load it, but was unable to, causing the "nan" issue. It should be fixed now. Let me know how it goes

svsergo, 2021-09-03:
Ok. Saving to memory works correctly.

svsergo, 2021-09-03:
An additional problem appeared during testing.  
[image]

closed 2021-09-03 by CAP1Sup