CAP1Sup/Intellistep #108: An additional problem appeared during testing.

CAP1Sup, 2021-09-03:
An additional problem appeared during testing.  
[image]

__Originally posted by @svsergo in https://github.com/CAP1Sup/Intellistep/issues/107#issuecomment-912640827__

CAP1Sup, 2021-09-03:
@svsergo what branch are you on? Could we stick to the dev branch for the time being, as it has the most fixes and updates? Is `USE_HARDWARE_STEP_CNT` enabled?

svsergo, 2021-09-03:
The firmware is loaded from the branch dev.  USE_HARDWARE_STEP_CNT - enabled.

CAP1Sup, 2021-09-03:
Alright, thanks for the info. Once I get some time, I'll add an option in the config_adv.h under the debugging section so that I can get the step count. If you wouldn't mind testing it for me, I could get a good idea of what's happening under the hood. Half the battle is figuring out what the heck is going on.

svsergo, 2021-09-04:
I am ready to test any firmware to achieve excellent performance.
Added the output to the com port:
#ifdef USE_HARDWARE_STEP_CNT
// Returns the count value of the timer-based step counter
int32_t StepperMotor::getHardStepCNT() const {
    Serial.println((TIM2 -> CNT) ); 
    Serial.println(stepOverflowOffset ); 
       return ((TIM2 -> CNT) + stepOverflowOffset);
}
float StepperMotor::getDesiredAngle() {
    #ifdef USE_HARDWARE_STEP_CNT
       Serial.println( getHardStepCNT()); 
        return (microstepAngle * getHardStepCNT());
The result of the work: 
1. original
      0
      0
      0
 
2.  steps 200 divisor 16
    623363 
    4294901760
    -3200
3. in the opposite direction steps 200 divisor 16
    65420
   4294901760
  -116

CAP1Sup, 2021-09-04:
It seems like there is a sensitivity issue with the hardware step counter. It doesn't always read correctly when counting down. That tells me that it's either an issue with the pull down / pull up resistor working incorrectly, causing it to count in the wrong direction. Notice that the error of the step count is even, as each missed step is counted in reverse. Would you mind doing a couple of test runs, checking to see if you can get an old step error? If you can't, that would mean my breakdown is correct, which would conclude the searching.

svsergo, 2021-09-05:
The number of missed steps is not repeated when counting in the opposite direction.
[test.pdf](https://github.com/CAP1Sup/Intellistep/files/7110841/test.pdf)

CAP1Sup, 2021-09-05:
When counting in the negative direction, is the step count error (the difference from where you know it should be and where it thinks) ever odd? I think that the direction signal is incorrectly configured, but I was curious if the issue ever appeared with odd step losses. Also, I can't access your PDF, it just says that something went wrong.

svsergo, 2021-09-05:
The error of counting steps occurs only when rotating in one direction. When rotating in the other direction, there is always no error.
[image]

svsergo, 2021-09-05:
I tested the firmware on another BTTS42Bv2 with an engine with a step of 0.9 degrees. The error is also present.

CAP1Sup, 2021-09-05:
Alright, I think that my analysis is correct. Try changing line 15 of the ` motor.cpp` file (currently `pinMode(DIRECTION_PIN, INPUT);`). Change the `INPUT` to either `INPUT_PULLDOWN` or `INPUT_PULLUP`. One of them will be correct. I think it should be `INPUT_PULLDOWN`, but I'm not sure. Let me know if that fixes your issue.

svsergo, 2021-09-05:
Both options have a negative result.  
The input signals are processed correctly. The motor rotates to the desired angle and the encoder shows the correct value.

svsergo, 2021-09-06:
When the filter is enabled  tim2ClkConfig. ClockFilter = 1, the number of pulses counts correctly. But the bill goes in one direction and does not depend on the DIRECTION_PIN signal.

svsergo, 2021-11-23:
After making changes, the error does not appear.

CAP1Sup, 2021-11-23:
>  After making changes, the error does not appear. 

It does appear very very slowly over time. I've even printed a part with no visible defects. Somehow, I can't get the step pin to recognize steps perfectly. It's so close but occasionally drops or adds a step. Check out the top issue. I'm going to close this for now as we have solved this problem.

closed 2021-11-23 by CAP1Sup