CAP1Sup/Intellistep #110: Readiness state?

arrowcircle, 2021-10-20:
Hey! I was watching this project and curious about it's state.
What features needs to be implemented to be production ready? Or at least usable?
Is project alive?

CAP1Sup, 2021-10-20:
As of right now, there is a bug where the hardware counter doesn't always register the backward step correctly. However, other than that, everything should be good. It's not ready for real world use yet due to that bug. There were some features that still needed work yet, but the basic functionality of movement should be more than adequate for real world use. I haven't been able to work on the project lately, but I'm hoping to get back into it again.

closed 2021-10-20 by arrowcircle

CAP1Sup, 2021-11-22:
@arrowcircle I actually just fixed the issue with the step counter not registering the direction correctly. There is only the minor issue of the very rare missed step, but I'll be working on that. The step correction isn't ready yet, but I think that for the first release I'm just going to stick with basic movement operations. I know that I haven't been active recently (school has been getting in the way), but I'm hoping to get back on this project, especially over the holidays.