CAP1Sup/Intellistep #113: Version 0.0.50 Steps seem off, OLED - locks up and will not update.

rob-mcinerney, 2022-01-19:
0.0.50 is the most recent version as of right now. Could you create an issue on my repository so that I have a note of your issue with the steps being off? Also, one for the OLED not updating? That would be helpful to organize things

CAP1Sup, 2022-01-20:
Hmm.. that’s really interesting. I’ll have to take a look into why the motor isn’t moving. Maybe it’s a configuration issue with the setup of the step pins? Just my first thought. I’ll try to get back to working on the firmware soon.

rob-mcinerney, 2022-01-21:
Yes I agree with the stepping, but that is a easy fix in the printer M92 Setup.

I did some more testing;
I can compile the code and get it downloaded just fine to the motor.
____________________________-Calibration_____________________________
On initial power up, it asks to be calibrated, I do that with Dip 4 ON, the motor moves one step (I can hear the click), waits for about 10 seconds and says it is done calibrating, maybe this is OK?  If I reboot then - like it says - the screen will go blank - but if I turn off Dip 4 - I can move forward.
I can change the divide by to 1/32 as long as it do it first. - More on that later.
_____________________Homing_____________________
The printer will home and jog just fine for a few minutes, then after about 3 minutes it locks up, I can't jog the motor (but it is still enabled) and the OLED display stops updating.  I do see that the temperature seems it go up to around 37 Degrees, then it stops updating and this is when the display locks up.
_________________Operation____________________________
If I reboot the motor (hard and/or soft) it will not respond, but if I wait a few minutes (maybe long enough for the temperature to come down???) it will re start and work for a few minutes again. 

Hope this helps and sorry for being long winded!

CAP1Sup, 2022-01-21:
No need to apologize for being long winded, it's great to have extra details. 

_**Calibration**_
The motor calibration is different then the original. Rather than calculating the overall distance moved, the calibration is only to get the encoder's offset to the full step. Basically, it's setting the zero of the encoder to the closest full step of the motor.

**_Homing_**
That's strange that it locks up. Try commenting `#define ENABLE_OVERTEMP_PROTECTION` in config.h (Add two slashes before it like this: `//#define ENABLE_OVERTEMP_PROTECTION`. I think that might be the issue. 

**_Operation_**
I think that this is caused by the overtemp. Try it without the `ENABLE_OVERTEMP_PROTECTION`. Just make sure that you don't overheat the chips next to the stepper motor connector (those are the main power controllers)

rob-mcinerney, 2022-01-21:
The devil is in the details!
Calibration works as advertised!
BTW - I really like the smoothness of the motor - MUCH better than the BTT code!

_Operation_
It is definitely a Temp issue, but disabling the protection didn't seem to help.  Here is the test I ran.
1. Compiled new code with //#define ENABLE_OVERTEMP_PROTECTION and I changed the software version number (just so I could know the changes made it into the .bin file)
2. Calibrated Motor and Homed Printer (motor stayed enabled - more on that later).
3. Watched the temp rise on the OLED, at about 35 degrees the OLED stopped updating and motor would not respond to move commands from Printer, Motor was still enabled (couldn't move shaft by hand).
4. Had to wait to get the temp back down - 5 minutes.
5. Rebooted and watched the OLED, for 2 reasons.
     - wanted to track the temp
     - wanted to watch the position (DEG:) on the screen so I could see when the OLED stopped updating, I moved the motor shaft by hand so I could see the DEG: changing.
6. With motor disabled temp rose to 33.1 and stabilized.
7. Enter heat gun!
8. Slowly heated up, temp - rose to 36.4 and OLED locked up
Additional info - If I load old bin file from BTT display and motor work fine (other than bad prints!) - just wanted to rule out possible hardware issues. 
Hope this helps in your troubleshooting.

rob-mcinerney, 2022-01-21:
Interesting addition - The STLINK is still active and able to report the CheckSum even when the OLED is blank....
But motor will not move is commanded by Printer.

CAP1Sup, 2022-01-21:
What about if you change the `SYSCLK_FREQ` in config_adv.h from 128 to 72? The system clock is actually only rated for 72 MHz, but I had no problem overclocking mine. Maybe that's the issue? The other thing is you could try disabling all the encoder code, but that's going to be a lot more complicated than a single line. I should have some time to add it as a debugging feature if you're still having issues.

rob-mcinerney, 2022-01-21:
Interesting enough, I tried that and noticed one other thing.  
After it locks up (which it still does at 72MHz), if it is rebooted (hard or soft) the "splash screen" (version number) and "Calibration / Flash loaded" screen display - then the OLED goes blank.  I can enable the motor and it will enable, but it will not move until it sets for a while.

CAP1Sup, 2022-01-21:
Hmm... that's really strange. I wonder what could be causing it. I pushed a commit that adds the `DISABLE_ENCODER` functionality in config_adv.h. Let me know how that turns out. I honestly have no idea why it would be locking up like that.

rob-mcinerney, 2022-01-21:
Uh-Oh!
I fat fingered the frequency, it is now set to 72 and OLED is still working, temp is over 42 degrees and still running.
Going to print test cube now...
Stay Tuned.

CAP1Sup, 2022-01-21:
Does the PID actually work? Are you using `ENABLE_PID`? What about layer shifting? That was the issue that I was seeing before. Would you mind sending a picture of the part you printed? Sorry for the large number of questions, I’m just curious as to your experience with the boards.

rob-mcinerney, 2022-01-21:
Print looks really good, Much better than stock BTT code.

This whole project started because Junction Deviation and Linear Advance could not stop my bad corners (much like others have posted).  I really looked like the printer was overshooting at the corner.  My old TMC2209's were better!
I am a old Industrial Servo Robot guy. 
So, I went looking for a way to tune the PID loops.  There was just no easy way of adjusting the gains, the BTT software wouldn't compile and I couldn't get their terminal program to run.

So,  found Intellistep and found it intriguing that there was a tuner.

But, I am not sure I can get much better than what I found so far.

But, I am willing to give it a go.
Haven't had a chance to look over the Enable_PID stuff yet.

I will try to get some photos, but the test prints are black (the PLA I had plenty of) and photos are hard to see. 
Do you think the Enable_PID is ready to be tested?

CAP1Sup, 2022-01-22:
Glad to hear that you got much better quality! I moved from TMC2130s to these on my X and Y because I thought that they would be better, but the performance is actually worse. The main issue is that the BTT one uses a full PID loop instead of processing the steps like a normal driver would, so the PID can lag behind or overshoot the set point (which is made worse by the fact that PID values are hard-coded in). That’s exactly why I started this project. I thought that the boards had potential, they just needed some better coding. I have it set now that the steps are processed immediately using processor interrupts, meaning that the steps are reproduced as closely as possible to what a typical driver would do. The added benefit is having the feedback from the encoder. I started to try to fix the BTT code, but A. That’s not even their production code and B. their coding uses absolutely no commenting and few libraries, so it’s very difficult to understand what is going on. I just went for a full re-write, which is what you see here. Unfortunately, due to trying to get basic functionality (such as stepping) working, some of the features (like PID) haven’t been updated and tested. I needed to build the foundation before starting to worry about the house. I’ll let you know when I get the PID functionality working. I’d love to see your prints, especially if you could put a piece of paper behind the part to help show the layers. I’m hoping that my miscounting issue is due to overclock, which would mean that I could release 0.1.0, marking the first complete release of the code. I’m not making a major release (like 1.0.0) until I have most if not all of the features in the config working. Thanks for your input!

rob-mcinerney, 2022-01-22:
Got it.  
Now that I look at what you have done (I looked further into your code further) and believe that what you have done is made the motor more accurate w/o implementing the PID.
I am using a Anycubic Mega S with many mods, the biggest is the Hot End - changed to direct extrude instead of the Bowden, added a drag chain to the X axis to control the cables, all of this added some mass to the head which I believe contributed to the overshoot.  I was hoping that the BTT S42B Servos (on X and Y) and their PID code could be used to tune the PID and therefore over come the additional mass - I like the direct extrude as I feel it does a better job of controlling the filament.

I think I that this will show that your code is better and was testing tonight with the speed at 100 mm/s and it looks better than the BTT code at 80 mm/s (their stock PID settings).  

I would be willing to run an A/B test on the same part (xyz calibration cube) as follows - (only changing the code on the X axis) ;
A. BTT stock Code at 100 mm/s - (I haven't run this yet) stopping the print mid print so the overshoot is easily identified, before the top layer covers it up.
B. Run your code at same settings 
I will use the same gcode file generated by PrusaSlicer (2.4.0).  For your thinking purposes and documentation;
- Acceleration at 1250 mm/s^2
- Speed at 100mm/s on perimeters, external perimeters at 50%
- Linear Advance - M900 K0.05
- Junction Deviation -  M205 J0.01
Your input on this test would be appreciated!  I can try it with white and black - to see which shows up better in a photo.  I can provide the gcode file also.

It is unfortunate that the BTT code can't compile right (so I can't hard code some PID changes) and I cant get the serial port to work and therefore adjust the PID on the stock code - if this worked I could do another comparison - your code and the BTT PID tuned.

Further testing - it might be good to tune the Linear Advance and Junction Deviation with your code.

Let me know if this test makes sense to you!
Thanks,

CAP1Sup, 2022-01-22:
Absolutely makes sense! I'm interested to see the results. I think that your issue is resolved now, so I would suggest we move to a discussion about the print quality of my code vs BTT's in the discussions section. I will add a link as soon as I set up the posting spot. Print quality observations that are marked with version and general settings will be great to have for comparison.

closed 2022-01-22 by CAP1Sup