CAP1Sup/Intellistep #119: Set up menu sequence - help

rob-mcinerney, 2022-02-08:
When setting up the motor some programming questions arise.
1. If you set the mA (it doesn't matter if the high current warning is displayed or not) when you use the K4 to confirm, the display exits to the main display (good news, the value is stored), thought it might be helpful to return to the menu screen after the K4 confirmation so you can continue to set up the motor, rather than exiting back to the main screen.
2. When you are setting mA or micro steps to a value that has the warning (high current or 1/32) - what does the statement "Back to Exit" mean, is it the K3 key that is supposed top be used to go back w/o saving?  If this is true, the documentation will cover that.
3. If you set the micro steps to 1/32, then exit, then set the mA to 1000, then exit, then go back into the micro steps it is switched to 1/16?
4. Can you explain the intent of the "saved config" menu selection?

CAP1Sup, 2022-02-08:
1. Check config.h. You can change `MENU_RETURN_LEVEL` from `MOTOR_DATA` to `TOP_LEVEL`. That should return you to the menu screen instead of the motor data.
2. I adjusted the "Back to exit" statement to "Back to cancel". I don't have my board to test, so I don't know if it'll fit on the OLED. Let me know if it overflows.
3. By switched to 1/16th, do you mean that the cursor is on 1/16th, or the actual set value is 1/16? The first is a display issue, the other is a whole menu issue.
4. The saved config is to modify the saved configuration. By saved configuration I meant settings saved in flash (so persistent between power cycles). Is saved config unclear?

Thanks for your feedback!

rob-mcinerney, 2022-02-08:
Got it,
1. I will have to compile a new version and try out 1 & 2.
2. For #3 - that is a good question, for sure the cursor is pointing to 1/16 and I assumed the actual set value was set to 1/16. but I didn't test it.  I assumed again that if the cursor was pointing to 1/16, that the actual value was 1/16.  Seems like I remember finding this because the motor moved the wrong distance when the printer commanded a move and I found the motor was set to 1/16 instead of the 1/32 that I need, but that was a long time ago.
3. For #4 - OK, I have a better feel for it now, I will have to investigate further.
Thanks

rob-mcinerney, 2022-02-08:
Update:
1 & 2 - I like setting the lave to TOP_LEVEL, that seems to work well, I can confirm that the "Back to cancel" will fit on the OLED, the documentation will cover this.
3. - If you don't save the configuration and reboot, you are right the mA and "Dir. Logic" revert to the default.  But, it seems that the "Microstep" stays where it was set, if I set it to 1/1 then after reboot is stay at 1/1 (it doesn't return to the default of 1/16).

CAP1Sup, 2022-02-08:
1 & 2 - Glad to hear it! I'll make sure to add it to the documentation.
3. Are you sure? There's nothing in the code that I can see that would cause the microstepping to be saved.

rob-mcinerney, 2022-02-09:
I will double check.  It is hard to get all the key strokes correct.
Stay tuned.

rob-mcinerney, 2022-02-09:
Update - I tried to duplicate the issue and am not able to duplicate at this time.

CAP1Sup, 2022-02-10:
Alright, I’m going to assume that it was a fluke. I think that after some minor documentation updates, I will be able to close this. Sound good @rob-mcinerney ?