CAP1Sup/Intellistep #124: CERN-OHL-S hardware license

dzid26, 2022-08-27:
I noticed the IntelliStep repo uses GPLv3 license at the root, so I assume it also applies to the Kicad files for any modifications. But GPLv3 is not a proper hardware license and doesn't protect design fully. 

Could you add this [LICENSE.txt](https://ohwr.org/cern_ohl_s_v2.txt) to [Kicad schematics](https://github.com/CAP1Sup/Intellistep/tree/master/Kicad/BTTS57Bv2) folder? This is the most "collaborative" [variant](https://ohwr.org/project/cernohl/wikis/faq#q-what-are-all-these-suffixes) of the CERN licenses.

I think @jlmxyz would have to approve👍.

CAP1Sup, 2022-09-08:
Are you sure you don't want this? I checked it a while ago but forgot to add it. Would you still like me to?

CAP1Sup, 2022-09-09:
All done!

closed 2022-09-09 by CAP1Sup