CAP1Sup/Intellistep #126: can't get the F* crc works.... need help

jlmxyz, 2023-04-01:
Hi!
I'm still working on this... I begun changes that made me far from what I expected to change... at first I just wanted to make it "true closed loop" eg have the encoder receive the order and drive the motor until position is reached... so I had to create a class to handle "absolute" position (rotation/angle) and perform operation on this....
so now, the encoder gets a position, and drive the motor in order to reach the position (since the motor in fact only know about 4 full physical steps position)... well it will when I will have finished (I'm planning also to implement "switch less" home/end detection, so that I can have a really absolute positioning system)

but I came into a problem with the UART design (since I use the UART to control the motor), so I moved the UART into a completly new design based on interrupts... which took me lot of time... and not fully debugged yet...

but the issue I have is communication with the encoder.... 
since the encoder is the key of a closed loop design, it should be safe to access it, so I need to rely on CRC and handle gracefully errors (system errors...)
so I re-activated the checkSafety, I had to fix also some code (the safety word was not read...)

so now, I communicate with the chip... that's a good point!
but CRC is almost always bad.... I sometimes managed to make it work, I've no idea if it's a side effect....

what I found so far : 
according to the doc, the CRC is computed on every byte of the message : the commands bytes, the writen bytes in case of write (yes I need write, since I need to be able to reset the chip in particular for the watchdog handling), the read bytes, and what I was not sure, the safety word....

I found using http://www.sunshine2k.de/coding/javascript/crc/crc_js.html that the whole safety word was used (something I find strange, because the CRC is computed using itself... but why not.... 

I also sometimes have period where the CRC is correct... by period I means I can have 60~80% success rate... letting me think that the computation of CRC is correct... I tried another source code found on https://github.com/fenugrec/freediag/blob/master/scantool/diag_l2_saej1850.c
that seems to give me same result.... 

but when the CRC is correct, I always fall into the error SYSTEM_ERROR.... so I tried to read the STATUS register to find what the SYSTEM_ERROR could be, but when I read the STATUS, I always get a CRC error ;) so I don't know if the bits I read are correct

and here I am.... I try to reset the chip, thinking it's a state issue...  but same, writing the reset create a bad CRC, reading the status create a bad CRC and reading the value register.... seems to give a good CRC.... 

here is some transaction I get when reading the status reg (offset 0) after a reset : 
`80 01 40 00 7F 15` : `80 01` is the command, which seems correct : read bit is set, addr is 0, 1 16 bit word is read.... `40 00` is the reply a bit odd... but... 7F is the first part of safety word, so flags are correct, a reset occured, no system error, no protocol error, angle is valid... and CRC isn't right... 

any idea?!?!!?!

jlmxyz, 2023-04-01:
ok... replying to myself... spending 4 days without a clue... and once you ask for help you find the solution....
1) RTFM https://www.infineon.com/dgdl/Infineon-Infineon-Angle_Sensor_TLE5012B-UM-v01_00-en-UM-v01_01-EN.pdf?fileId=5546d46146d18cb40146ec2eeae4633b
2) ask google https://community.infineon.com/t5/Community-Information/TLE5012B-CRC-calculation-trouble/td-p/296024
3) the CRC is not computed on the safety word (which is a flaw, it should at least include the 1rst byte of the safety)
4) check if crc is correct http://www.sunshine2k.de/coding/javascript/crc/crc_js.html without the safety word
5) yes it is... retry... yes it is....
6) why my buffer is half filled?
7) why is my command word's length half the size?
8) ohhh! shit, it use length in byte words not uin16_t words!!!
ok... now working...
and reset is working also....

closed 2023-04-01 by jlmxyz

CAP1Sup, 2023-05-15:
@jlmxyz sorry, I've been busy finishing up the college semester and I'll be working over the summer. I'm glad to hear that you figured out your issue. How is the project going currently? Do you have any other questions?

jlmxyz, 2023-07-09:
@CAP1Sup sorry for late answer, didn't found time to work on it lately, just got some motivation back!

ok it... was... working... and now... no more.....  :/

I have added a reset of chip it encoder object constructor, so far nothing that could be wrong... the SPI interface is configured, there (sadely) no way to power on/off the encoder... 

what I see it that I get some communication on reset the whole reply transaction is 
0x00 0x11 0x18 0xEF 0xBF 0x14 0xFF 0xFF 0xFF 0xFF
so write, lock=0, upd=0, @1, ND=1
data sent 0x18EF (big endian) (accord to doc reset value is 0x18EE, so only AS_RST is set to 1 comparing to reset)
STAT = No reset, system error, no interface access error, angle valid
RESP = 1111
CRC = 0x14
remaining 0xFF is padding I use to detect if there something wrong on the line
according to http://www.sunshine2k.de/coding/javascript/crc/crc_js.html CRC8_SAE_J1850 the result must be 0x29, computed on 0x00 0x11 0x18 0xEF, which is also what is computed by the CRC function....

so CRC is bad.... the same reply occurs on every retries.... so that's not a "issue not reproductible"
last time it was the same....  somehow it started working last time, but this is really a problem, in particular if this is the chip reset that create the issue....

I added also a read of status register, just to check if it was a issue with writing and what the status of the chip
0x80 0x01 0x40 0x00 0xFF 0x1B
accord to the website CRC should be 0x09....
crc fail or running? seems reproductible on this run (was not in yesterday tries...) how to reset the eeprom???

CAP1Sup, 2023-07-14:
@jlmxyz I remember having all kinds of issues with the encoder. BTT designed the board wrong… that’s why the library from Infineon doesn’t work. I was able to extract some of the work arounds that BTT made, but I don’t know if they work perfectly every time.

I started using Klipper on my A8 of Theseus and I think that there’s a possibility that Klipper could work with these boards. There would have to be a whole interface written to map the step commands to the stepping functions in this repo… a virtual end stop is another possibility for stall detection. Klipper sends digital step counts, so it would be easy to accumulate them into the desired step position. Maybe if I have time, I’ll try. I really doubt that I will have time… and even if I do, the chances of everything working… not high. I did notice that BTT released a newer version of the board, so maybe they fixed some of the issues… I’ll probably just stick with my TMC2130s. If I do begin working on a Klipper version, I’ll let you know. Best of luck and I’m here if you need anything… feel free to message anytime.

jlmxyz, 2023-07-14:
Hi!
thanks for the encouragements!

if you work on klipper, if you can figure how the protocol is built it would be awesome! I couldn't find any doc, that's why I ended directly communicate with gcode commands.... adding some custom for time shift / sync....

for the encoder, I noticed that the main tree of intellistep is communicating nicely with the encoder where my changes fails... so trying to figure what settings cause the error....

next will be the smart pointer messaging test (interrupt handler uart/can allocate commands managed by smartpointers, so that the message can be passed across the handlers and forwarded/replied to a communication interface, and for uart, shall now work using "silence" (a time of a byte transmission, without any transmission is reported by hardware as a silence in communication) to detect end of message sending(now checking is last received char is a '\n' if so, the buffer is parsed to extract the commands)) this first step to avoid copies/interrupt handling of messages instead of char by char handling and memory allocation... next step will be to have commands string_view... so that parser that mainly just split commands will avoid lot of string allocations.... this shall save lot of cpu.... and memory

but now, detect the diff in configuration for the communication with the encoder...
best regards

-------- Message d'origine --------

CAP1Sup, 2023-07-17:
I think that the best route would be adding some of modified code from this repository into a branch of the original Klipper firmware. That way, you wouldn’t have to deal with any protocols or timing… you could just focus on adapting the stepping functions to work within Klipper. Also, if Klipper ever updated its protocol, it would be as simple as pulling from the upstream branch to get it working again. Another benefit is being able to use the other pins of the board for any other Klipper supported function/component. Like I said… it’s something I might work on, but unfortunately I think that I’m going to be too busy with college. Best of luck with your project… sounds very involved!

jlmxyz, 2023-07-18:
the problem is that klipper is designed as "step/dir" interface, the motor controllers are "dummy" slaves, eg gpio for example are controlled by the klipper master (ex the master send some command like "put gpioA OD") so it's not designated to be smart... so no closed loop... no smart behavior...in fact klipper is not "autonomous" system talking together, put more a "set of boards that looks like one"

-------- Message d'origine --------

jlmxyz, 2023-07-28:
seems that I fixed it??? until... 
after lot of time and tries, something bothered me... whatever my changes, the crc was always the same... it was too reproductible to be hazard... so yesterday I decided to give a try to the computation of crc using the checkup table of datasheets/Infineon-Angle_Sensor_TLE5012B-UM-v01_02-en-UM-v01_02-EN.pdf p41.... and it worked... and today it still works... 
so my guess is that the TLE5012B is *NOT* compliant with CRC8_SAE_J1850 but they instead used the checkup table they provided in doc and for some reason the checkup table is wrong.... but with the same error on both side, it end working... 

hoping it will be the last time I "fix" this.....

CAP1Sup, 2023-07-28:
BTT made a design error and didn’t include a resistor recommended in the datasheet (if I remember correctly), meaning that the pin has to be reconfigured every time you want to read/write to the TLE5012B. I think at this point I’m going to sign off of the project. Between BTT’s design errors and [the salmon skin effect](https://github.com/Klipper3d/klipper/issues/3103#issuecomment-660684436), it doesn’t seem like there’s a way to get these boards working to the quality of something like a TMC2130. Best of luck with your pursuit and let me know if you have any questions!

jlmxyz, 2023-07-28:
no, the resistor is present, even if not the right value... 680ohm instead of 470 in datasheet... if one day I have time, I will replace the resistor to see if this fix the issue.... since we have a almost working design it should be not far from doing a full opensource solution!

jlmxyz, 2023-07-28:
BTT didn't made so bad design, maybe some error like the resistor, but compared to makerbase that didn't included optocouplers to avoid electrical issue, and the fact that there are 2 high speed interfaces that allow one board to act as a master of another... (for example the master receive position info on CAN, and drive a slave using USART...) the board can also easely be modified to add temperature sensor, power temperature driver or autoleveling probe.... well, it's a good base to create a perfect board, if we can manage to make it work

CAP1Sup, 2023-07-28:
I wouldn’t recommend replacing the resistor… I think that the board will work fine as long as the GPIO reconfiguration hack is consistent. I am interested in the CRC work you’re doing… I noticed that sometimes the encoder sends back a bad reading, which is why I had to implement such aggressive averaging. I did take a look into the salmon skin effect… it’s caused by [the drivers not being able to output correctly near the zero crossing of the sine wave.](https://cabristor.blogspot.com/2015/02/drv8825-missing-steps.html?m=1) Maybe it could be fixed in software… or maybe the redone stepping algorithm already fixes it. I don’t currently have access to an oscilloscope, so I have no way to test the drivers. I will be returning to college soon, so theoretically I could do testing, but I doubt that I’ll have time. I do agree that these boards have a lot of potential… they could be used for a number of things. I feel as though Klipper could be adapted to run the boards in at least open loop mode… the stepping function shouldn’t need timers besides the PWM, which can be done the same way that PWM fans and heaters are currently done

jlmxyz, 2023-07-30:
> I wouldn’t recommend replacing the resistor

this would be a test for a next pcb design (this time open source :) ) 

> I am interested in the CRC work you’re doing

I'll rebase onto v0.1.0 and push my branch once I've done sufficient tests (last time I got it working, I stopped working on the project for a couple of months and when I came back crc was not working again...)

>it’s caused by [the drivers not being able to output correctly near the zero crossing of the sine wave.](https://cabristor.blogspot.com/2015/02/drv8825-missing-steps.html?m=1)

does this apply to A4950E? it seems also that the effect he sees is caused because the motor is a 3.3V motor so that the bug is more seeable, the fix he propose is to increase the motor voltage by adding some diodes so that the problematic zone is not used, but BTTS57Bv2 is designed for motor at least 24V... anyway there are good information about the noise, which is really problematic, will check what mode the drivers are in (fast, slow or mixed decay) and see if switching is necessary

jlmxyz, 2023-08-01:
I suspect more the salmon skin effect caused by the use of float computing almost everywhere... 
the issue with float is that it's not evenly distributed, which means that between too successive float value the "distance" change where int computation is evenly distributed, this can lead to some big issue 
sure, float is convenient to develop with, you don't have to transform the values and says "16384 is 180" "0 is zero" "32767 is 360-1/2^15)" but they can create salmon skin effect (thinking it is at right position when in fact it should be somewhere between the 2 successive float value)

CAP1Sup, 2023-08-01:
I'm not sure about your claim of floats causing the salmon skin effect, but I could see how it could cause such an issue. Intellistep uses integer step counts for all motor calculations and a simple sine lookup table. Ironically, Klipper also uses a similar integer based step counting, which is why I believe that a Klipper integration might be possible. Before I commit to integrating the code in this repo into a branch of Klipper, I'd like to do some testing with an oscilloscope to see the difference. If there is a dead zone in the original firmware and Intellistep fixes it, then I will consider integrating some of Intellistep's code into Klipper. Regardless, if I do testing (maybe around a month from now), I'll let you know the results either way.